RobTillaart/PCF8574 #2: Read interrupt callback

poky, 2020-12-07:
Hi, thank you for the library, just wondering is there a read interrupt callback function that I can set a int pin to read when there's a interrupt on the PCF8574, thanks!

RobTillaart, 2020-12-07:
@poky 
No there isn't. 

It would be similar to **attachInterrupt()** and as that already exists I do not "copy" that into the library.

That said it might be a good idea to have an example that shows how the INT pin can be used. 
Will give it a try today.

RobTillaart, 2020-12-07:
Can you test this example?
It sets a flag when a line of the PCF changes
As I have no appropriate HW nearby I cannot test it right away :)
thx.

```cpp
//
//    FILE: PCF8574_interrupt.ino
//  AUTHOR: Rob Tillaart
//    DATE: 2020-12-07
// PUPROSE: test PCF8574 library
//

// TEST SETUP
// Connect INT pin of the PCF8574 to UNO pin 2
//
// (from figure 4 datasheet
// Place a pull up resistor between pin and 5V
// Place a capacitor 10-400pF between pin and GND

#include "PCF8574.h"
#include <Wire.h>

PCF8574 PCF(0x38);

////////////////////////////////////
//
// INTERRUPT ROUTINE + FLAG
//
const int IRQPIN = 2;

volatile bool flag = false;

void pcf_irq()
{
  flag = true;
}

////////////////////////////////////
//
// MAIN CODE
//
void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("PCF8574_LIB_VERSION: ");
  Serial.println(PCF8574_LIB_VERSION);

  PCF.begin();

  pinMode(IRQPIN, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(IRQPIN), pcf_irq, FALLING);
}

void loop()
{
  uint32_t now = millis();
  if (flag)
  {
    flag = false;
    int x = PCF.read8();
    Serial.print("READ:\t");
    Serial.print('\t');
    Serial.print(now);
    Serial.print('\t');
    Serial.println(x, HEX);
  }
  // do other things here
  delay(10);
}

// -- END OF FILE --
```

RobTillaart, 2020-12-07:
Example is merged into 0.2.2 version of the library

poky, 2020-12-08:
Got it to work, thanks for the heads up!

closed 2020-12-08 by poky

RobTillaart, 2020-12-08:
You're welcome