RobTillaart/PCF8574 #6: ambiguous readButton8 overloads

sonatique, 2020-12-14:
Hello!
Arduino compiler complains that call to readButton8() is ambiguous, and I think he's right:

From PCF8574.h lines 44 and 45:
`inline uint8_t readButton8()  { return PCF8574::readButton8(_buttonMask); }`
`uint8_t        readButton8(const uint8_t mask = 0xFF);`

How could the compiler knows whether to call readButton8(0xFF) or readButton8(_buttonMask) then?

I would suggest removing "= 0xFF" at line 45.

RobTillaart, 2020-12-14:
Thanks for pointing to this ambiguity. I will try to fix it today.

RobTillaart, 2020-12-14:
Traced it back to a PR in version 0.1.07, seems to be seldom used.

Very recently added CI and unit tests in this repo. Tests are at minimal level and need to be extended.
Apparently the examples did not have/ show this ambiguity.

sonatique, 2020-12-14:
Hello, you're welcome. Thanks for your very prompt action, and of course thanks a lot for providing this library in the first place, so useful and well done!

RobTillaart, 2020-12-14:
You're welcome, if there are other remarks let me know.
Even if there are new libraries you might need, feel free to ask. 
If time permits and testing is possible I can write them.

closed 2020-12-14 by RobTillaart

sonatique, 2020-12-14:
Thanks a lot, amazing! I am just starting a project with PCF8574, so no other big issues so far.
I have to admit that the use case requiring the use of readButton8 vs read8 in the case of inputa only (push buttons only connected) is not 100% to me after reading the documentation: I understand that readButton8 is useful when we want to achieve "pseudo-bidirectionnal" on one pin, but somewhere else on the internet I read that one need to shortly output high before reading in order so somehow achieve pull-up (if there are no physical pull-up resistor in place), thing that is done by readButton8.
I guessed using readButton8 instead of read8 inside my interrupt handler won't harm, but I also observed that simple read8 seems to give correct values everytime, so well. I sticked to using readButton8 for "safety" but I am not sure whether it is useful or not at all (I tend to prefer to understand everything and use minimal working solution instead of cargo-cult, but well, sometimes I am too lazy ;-) )

So well: if you could complete readme for readButton* with an indication of use cases, or benefits over read8, I think it would be useful for me and others, but that is not a big deal, one can experiment by ourself, you already saved us a lot of time an headaches with your library!

I also noted a reference to an unknown / unfound "figure 4 datasheet" in https://github.com/RobTillaart/PCF8574/blob/master/examples/PCF8574_interrupt/PCF8574_interrupt.ino but this is really minor of course.

Thanks again!

RobTillaart, 2020-12-14:
The **readButton()** code was from a 3rd party and (s)he provided 2 examples.  I never used them myself outside these examples.
To update the readme for these, I first must understand them better, as no documentation is less worse than wrong documentation.

As I am in the middle of upgrading all my libraries to include Arduino-CI (Continuous Integration) updating readme's have relative low priority. That said please open an issue with this documentation request so it won't get lost.

For the datasheet, please open an issue for that too as there are many PCF datasheets from different suppliers.
And yes they all have a different figure 4  ...

sonatique, 2020-12-14:
Done, hopefully to your satisfaction. Thanks

sonatique, 2020-12-17:
Hello Rob,
From what I can see by looking at your last commit aimed at fixing the issue (and going to version 0.2.3) https://github.com/RobTillaart/PCF8574/commit/159923e2c7a766c294543c3d260ad0a9ec852625
you only removed the "= 0xFF" in the readme file, not in the actual PCF8574.h header file...

I think this was not intentional, but I would say the issue is not fixed in master, right?

RobTillaart, 2020-12-17:
Ouch, yes my mistake. WIll fix it right away

RobTillaart, 2020-12-17:
Fixed, I recreated my 0.2.3 release, so if you download that (or revert to 0.2.2 and back again in lib manager) 
than it should be fixed. Please let me know if it works now as expected.

Sorry for the inconvenience.

sonatique, 2020-12-17:
Hello Rob, thanks again, no worries!

sonatique, 2020-12-17:
Hello Rob,
I just did revert to 0.2.2 and back to 0.2.3 (and verified that files are changed), but the file I got for 0.2.3 is still not the latest good one.

Maybe there is a cache holding the previous files somewhere, I don't know, but it is not a big deal for me right now.

RobTillaart, 2020-12-17:
Manually throwing away the library?

sonatique, 2020-12-17:
Same. The previous version seems to be cached somehow from where the lib manager downloads it.

RobTillaart, 2020-12-17:
Windows? Check C:\users\name\appdata\ ?

sonatique, 2020-12-17:
No way. I delete everything I could, I keep on getting the original 0.2.3 header file...

RobTillaart, 2020-12-17:
https://forum.arduino.cc/index.php?topic=528210.0 ?

sonatique, 2020-12-17:
I don't have such arduino_xx folders there...

RobTillaart, 2020-12-17:
Check under File > Preferences > Show verbose output during > Compilation (check) > OK 
to see the path of the libraries that are compiled in the black console window at the bottom of the Arduino IDE window.

sonatique, 2020-12-17:
I checked from where the compiler take the library files (its from /xyz/Arduino/Library/PCF8547 ). When I revert to 0.2.2 its those files that change. When I udpate again to 0.2.3 its also those files that changes. But they changes to the old 0.2.3 not to the new one. I am pretty sur that the wrong old file is downloaded.

I completly removed the folder, deleted all temps and cache file, but still, when downloading 0.2.3 I get the old version, not the one from today.

I would suggest to change version 0.2.4 because I might not be the only one affected

RobTillaart, 2020-12-17:
YOu have a point. I'll add another release.

sonatique, 2020-12-18:
0.2.4 is fine, thanks!

commit ef9d989f (2020-12-14): Develop (#8)

* fix #6 ambiguity readButton8()
* update version to 0.2.3
commit a5cf513d (2020-12-17): fix #6 again (#12)

- redo 0.2.3 release too
commit d4a57f31 (2020-12-17): Develop (#13)

* fix #6 tag 0.2.3 => 0.2.4