RobTillaart/PCF8574 #14: Support for Wire1/Wire2/Wire3/Wire4/Wire5

mattbue, 2021-01-02:
Hi, i have modified your code to support Wire1, Wire2, Wire3, Wire4 and Wire5 interfaces.
Are you interested on it for implementing into your code?

RobTillaart, 2021-01-02:
Yes, please, if you can make a pull request I can review it.

For which platform do you use it? ( just tech curiosity )

mattbue, 2021-01-02:
Great. Request is comming the next days.
Currently the my code runs on Teensy 4.1 using Wire2 interface.

RobTillaart, 2021-01-02:
Please include the following
- [ ] update version in library.properties  -  0.2.5 or 0.3.0 )
- [ ] update version in library.json
- [ ] update version in .cpp and .h file 
- [ ] update the history in .cpp
- [ ] update the readme.md
- [ ] update license to include 2021 (Happy New Year!)
- [ ] ...

mattbue, 2021-01-03:
I just created the pull request. But just included the .cpp and .h file.
Let me include the other stuff.

So I'm closing this issue.

closed 2021-01-03 by mattbue