RobTillaart/PCF8574 #17: Can't create multiple devices

LorenzoDelCerro, 2021-03-18:
When trying to call the constructor twice, for two PCF8574A devices with different addresses : 
[image]
[image]

I get the following error from the arduino compiler :

[image]

I couldn't find an example with two devices, so here I am !

LorenzoDelCerro, 2021-03-18:
My bad, somehow the library manager duplicated the file, thus the dual include... Sorry for being a moron and not seing this earlier !

closed 2021-03-18 by LorenzoDelCerro

RobTillaart, 2021-03-18:
No problem, happens to all of use some day