RobTillaart/PCF8574 #18: work with tm1637

alirezaimi, 2021-04-16:
Hi
I use xreef library and can't work with tm1637 7segment module, 
Is there any luck in using this lib ?

RobTillaart, 2021-04-16:
Hi,
I don't know the xreef library so I cannot comment on it.

> Is there any luck in using this lib ?

There are many different modules, so just give it a try as that is the real proof it works with your module and code.

RobTillaart, 2021-04-16:
But be aware the PCF8574 is a complete other chip than the TM1637.
So do not expect it too work.... 
(but then again I do not know anything about your HW setup, code etc)

RobTillaart, 2021-04-16:
I have a draft TM1637 lib that is not tested well enough or published yet. 
Can setup the repo as experimental...

closed 2021-04-16 by RobTillaart