RobTillaart/PCF8574 #20: Platformio contains outdated package

thijstriemstra, 2021-04-23:
When I install this library using platformio it downloads and installs 0.2.1 instead of 0.3.0

See releases on https://platformio.org/lib/show/1358/PCF8574

thijstriemstra, 2021-04-23:
A temporary workaround would be to add this to `platformio.ini`:

```
lib_deps =
  ; use master, see https://github.com/RobTillaart/PCF8574/issues/20
  PCF8574=https://github.com/RobTillaart/PCF8574/archive/refs/heads/master.zip
```

RobTillaart, 2021-04-23:
Sorry I do not maintain the platformio archives.  A quick look shows that platformio misses multiple releases of the library.  
Probably there are more of my libraries not up to date. I need to go through my mail to find who was in charge then.

Thanks for pointing out, to be continued.

ivankravets, 2021-04-23:
See https://github.com/RobTillaart/PCF8574/pull/21

"/" is not allowed char in a keyword.

RobTillaart, 2021-04-23:
@ivankravets 
Is there a check I can add in my actions to prevent such error (like I have a json checker)?
See - https://github.com/RobTillaart/PCF8574/actions

RobTillaart, 2021-04-23:
#22 in progress

RobTillaart, 2021-04-23:
#22 merged, problem should be solved by know

@ivankravets 
Thanks for your fast and kind support

@thijstriemstra 
Thanks for reporting the issue. 
Can you please verify that the problem is solved?  If needed reopen this issue!
It might take some time before it is available in PlatfromIO

RobTillaart, 2021-04-23:
Will apply a fix to other libs that have I/O in it e.g. PCF8575
(A productive Friday evening :)

closed 2021-04-23 by RobTillaart

RobTillaart, 2021-04-23:
PCF8575 done => 0.1.1 released

RobTillaart, 2021-04-23:
MCP23017_RT done => 0.2.2 release

@thijstriemstra 
PCF8574  version 0.3.1 is available - https://platformio.org/lib/show/1358/PCF8574

RobTillaart, 2021-04-23:
No other '/' in keywords field found. 😊

ivankravets, 2021-04-23:
> Is there a check I can add in my actions to prevent such error (like I have a json checker)?

Yes, you can do this with `pio package pack` command. This "checking" was added in the latest development version.

You need the next lines in your CI configuration file

```
pip install platformio
pio upgrade --dev
pio package pack
```

Does it work?

thijstriemstra, 2021-04-23:
@RobTillaart yep, thanks!

RobTillaart, 2021-04-23:
@ivankravets 

Pip install etc does work, published several libraries this evening!  (nice tools)

thanks again,
Rob

ivankravets, 2021-04-24:
We don't have some libraries in the registry. We don't import libraries automatically. Developers should manually publish their packages and new updates.

For example, `PCF8575` library is not listed in the registry. You need manually to publish it using `pio package publish` command. See https://docs.platformio.org/en/latest/core/userguide/package/cmd_publish.html

RobTillaart, 2021-04-24:
Yes, I succeeded yesterday to get all up and running. Need some time to see how to integrate it into my way of working.

ivankravets, 2021-04-24:
If you mean to automate this, you can generate https://docs.platformio.org/en/latest/envvars.html#envvar-PLATFORMIO_AUTH_TOKEN and pass it to Github CI. Then "on release" event just call `pio package publish --non-interactive`