RobTillaart/PCF8574 #24: Question - constructor in setup() function?

ear9mrn, 2021-05-14:
Hi there,

I would like to use an I2C scan to find the address of my module then run the constructor. As far as I can tell this would have to be done inside the setup() function (scan then constructor) and if I do this the constructor is then not visible outside the setup() function as it would not be a global variable. Any suggestions as to how I could do this (i.e. not know the address of the module until the setup() function is run)?

Thanks,

Pete.

ear9mrn, 2021-05-14:
so I think I have figure it out.

```
PCF8574 *pcf8574;

setup() {
     pcf8574 = new PCF8574(0x27);
     pcf8574->pinMode(0, INPUT);
}

loop() {

         Serial.println(pcf8574->digitalRead(0));

}

```

RobTillaart, 2021-05-14:
something like this?

```cpp
#include "PCF8574.h"

PCF8574 PCF_01(0x38);  // just give it an address as placeholder

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.print("PCF8574_LIB_VERSION:\t");
  Serial.println(PCF8574_LIB_VERSION);

  Wire.begin();

  // knowing the address is in the range 0x20..0x3F
  bool found = false;
  uint8_t address = 0x20;
  for (; address <= 0x3F; address++)
  {
    PCF8574 test(address);
    if (test.begin())
    {
      found = true;
      break;
    }
  }

  if (! found)
  {
    Serial.println("address not found, check hardware and reboot");
    while (1);
  }
  else
  {
    Serial.print("found\t");
    Serial.println(address, HEX);
  }

  PCF_01 = PCF8574(address);
  PCF_01.begin();
}

void loop()
{
  Serial.println(PCF_01.read8());
  delay(100);
}

// -- END OF FILE --

```
It will trigger on the first address found, not necessary an PCF8574 . 
most I2C devices do not have a signature / ID register.

An elaborated model can scan the whole range and will indicate the addresses found. 
it would be better to use core Wire commands (code of the isConnected function) to find the address 
and not create N PCF8574 objects.

Hope this helps.

RobTillaart, 2021-05-14:
> so I think I have figure it out.
> 
> ```
> PCF8574 *pcf8574;
> 
> setup() {
>      pcf8574 = new PCF8574(0x27);
>      pcf8574->pinMode(0, INPUT);
> }
> 
> loop() {
> 
>          Serial.println(pcf8574->digitalRead(0));
> 
> }
> ```

With a pointer one does not create the dummy object. Think that is better,👍

closed 2021-05-14 by ear9mrn