RobTillaart/PCF8574 #25: Empty constructor for setting the device address later

alve89, 2021-07-03:
Hi Rob,

since I'm using your library within one of mine, I needed to be able to set the device address within my class and so not directly while calling the constructor. As I thought this could be helpful for other people as well, here you have my edited files:

1. Adding a default address (0x20, all address pins are connected to GND)
2. Adding an constructor with no address parameter
3. Adding a function `setDeviceAddress()`

[PCF8574_emptyConstructor.zip](https://github.com/RobTillaart/PCF8574/files/6758299/PCF8574_emptyConstructor.zip)

RobTillaart, 2021-07-03:
Looks like a good addition. Might take some time as I am busy. 
Think you only need to overrule the address, so no extra constructor needed.

RobTillaart, 2021-07-04:
@alve89 

Added develop branch that implements
```cpp
  bool    setAddress(const uint8_t deviceAddress);
  uint8_t getAddress();  
```

You could use a standard constructor with a dummy address and overrule this with **setAddress()**

This function returns true, if this address can be found on the I2C bus as I though that allows you to "scan" a range or list of possible addresses to find the right one.  No extra examples as it is quite straightforward.

The function **getAddress()** is for convenience and allows some (to be written) unit tests.

Can you verify this works for you?

commit 11e4a24d (2021-07-05): fix #25 add setAddress() (#26)

* fix #25 add setAddress() + refactor
closed 2021-07-05 by RobTillaart