RobTillaart/PCF8574 #27: How to connect multiple PCF8574 devices using address from scanned procedure

paulsteigel, 2021-07-21:
Hi Mr. RobTillaart and others,
Great thanks to your valuable contribution to the creation of this library. I am a newbie and currently using ESP8266-01S (MCU) to control a range of several pcf8574 chips via I2C (using GPIO1, GPIO3 on ESP8266), Interrupt pin of pcf8574s are connected together to GPIO0 on the MCU. I have gone thought your documentation and examples but I did not see a way to connect an I2C device based on the address obtained from I2C scanners. All the known examples was to hard-code slave addressé with constructor rather than using a dynamic slave address initialization.

Could you please be so kind to guide me on this part with thanks. Please see my issue as described below:
1. MCU (ESP8266) will scan the bus every 10-60 minutes or so to detect new plugged in Pcf8574 module;
2. If new device is found, MCU will store the address of the new device on EEPROM or a config file using LittleFS
3. MCU will initialize a new connection to the new pcf8574 device with the new address and conduct check on status of its pin if an interrupt signal was detected at GPIO0 (of course a read request will have to be issued to every Pcf8574 devices to find which device has a pin with status changed).
Thank you very much for your time and attentions. Please advice me and reach me at ngocdd@sfdp.net with thanks.
yours very faithfully,
Dang Dinh Ngoc

RobTillaart, 2021-07-21:
Note: this is more a question for the Arduino forum.

>  All the known examples was to hard-code slave addressé with constructor rather than using a dynamic slave address initialization.

The library is meant to support hard wired devices only. 
Every device needs an unique address to be set by the address pins. (16 different addresses possible)

- Make a function that gets the found address as parameter and use that address to create a temporary object within that function. Make the settings you want and the device will hold the values.. (disclaimer never tried)
- Create all possible 16 objects, put them in an array and you can do anything you want. (disclaimer never tried)
- Why you want to store the addresses? Can devices disappear after popping up?
- Your setup cannot detect which PCF did trigger the interrupt. Why not add a PCF8575 with 16 IO lines to connect the IRQ lines to. The IRQ of the 8575 is connected to the IRQ of the ESP.
If a 8574 device triggers an interrupt, the 8575 will forward that interrupt and you only need 2 reads (1x 8575  1x 8574)

paulsteigel, 2021-07-21:
Dear Mr. RobTillaart,
Thank you very much for your quick response. I checked on the arduino
forum, your topic was in 2012 and has been closed, so I would like to post
on Github. For this, please be sorry for bothering you.
Currently, I am designing a SSR relay with pcf8574 on each
relay. This relay can be plugged in an I2C bus (support upto 8 slaves) and
an additional Interrupt line to join all INT of the plugged in relays
(wired up to GPIO0 of the main MCU)
Relay address is set by a 3 bit DIP switch.
In the relay design, a touch button will be connected to P0 pin, and P7 is
the output pin to trigger the relay ON/OFF. The whole job of the I2C (plus
the INT rail) is to allow the MCU to know when to query the I2C slave (1-8)
based on the signal received on GPIO0. I checked the NXP application note
and it allows connecting PCF8574 INT together.
I have to use this approach to cut the cost of the entire hardware.
I looked through your code and found that it can be modified to suit my
needs. What I am trying to do for now is develop a I2CBus class with few
features as below:
1. Scan (SDA, SCL): to scan for new devices
2. Connect (address) to Connect to new device
3. Read (Address, Pin number)
4. Write (Address, Pin number)
So it will work that way! I hope I can make it shortly despite the fact
that I am not quite familiar with C++ (your code was very educational to
me).
Again, Thank you very much for your attention and guidance.

Dang Dinh Ngoc
Freelance consultant on M&E and Governance

RobTillaart, 2021-07-21:
You can always create a new topic on the forum.

What you are asking for is outside the scope of this library.

One question:
How much devices can be connected simultaneously?

if the answer is 1 it is rather easy, if it is more than 1 you need an administration of addresses that are online, offline and changed. No rocket science but not trivial either. could take a day or so to create including enough testing.

paulsteigel, 2021-07-21:
Thanks for your reply, I will create a topic on the forum during the creation of the procedure then. 
Regarding your question: "How much devices can be connected simultaneously?"
I have from 1 to 8, in most case shall be 6 (as the entire MCU Power adapter 

- [ ] 

- [ ] and also relay will be installed in a wall plate for controlling in house light).

Scanning function will tell the online/offline status of an I2C device.

RobTillaart, 2021-07-21:
Simplest way is to allocate an array of 8 devices that always exist in software.

Scanning will be going over the array and calling the **isConnected()** function, and depending on the return value true or false and the known previous state one can decide if it is still online, still offline or changed from one to the other.

RobTillaart, 2021-07-22:
@paulsteigel 

you might take a look at this code snippet (**disclaimer: code is not tested**)

it creates 8 PCF objects in an array
it has an array of 8 elements to hold the last known status per PCF

in loop it scans all 8 of them at a time interval (you might change that to an interrupt triggered flag)

depending on the newState and the lastState an appropriate action can be taken. 
(now mainly comments, but you should get the idea)

```cpp
//    FILE: PCF8574_array.ino
//  AUTHOR: Rob Tillaart
//    DATE: 2021-07-23

#include "PCF8574.h"

PCF8574 P0(0x38);
PCF8574 P1(0x39);
PCF8574 P2(0x3A);
PCF8574 P3(0x3B);
PCF8574 P4(0x3C);
PCF8574 P5(0x3D);
PCF8574 P6(0x3E);
PCF8574 P7(0x3F);

PCF8574 PCF[8] = { P0, P1, P2, P3, P4, P5, P6, P7 };
bool lastState[8];

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);

  for (int i = 0; i < 8; i++)
  {
    lastState[i] = PCF[i].begin();
  }

  Serial.println("exit setup");
}

void loop()
{
  for (int i = 0; i < 8; i++)
  {
    bool newState = PCF[i].isConnected();

    if (newState == false)
    {
      if (lastState[i] == false)
      {
        // it was disconnected and stays disconnected => no action
      }
      if (lastState[i] == true)
      {
        // it was connected and became disconnected => possible action?
      }
    }
    if (newState == true)
    {
      if (lastState[i] == true)
      {
        // it was connected and stays connected => possible action?
      }
      if (lastState[i] == false)
      {
        // we have a change so do the read and write here
        // e.g
        int x = PCF[i].read8();
        Serial.print(i);
        Serial.print('\t');
        Serial.print(x);
        Serial.println();
        // ... whatever
      }
    }
    // update last known state
    lastState[i] = newState;
  }

  // other things here
  delay(1000);
  Serial.println(millis());
}

// -- END OF FILE --

```

paulsteigel, 2021-07-22:
Dear Mr. RobTillart,
I would like to thank you a million for your time and effort on this.  I will take your code as the entry point for testing and make it further developed to fit with my need on I2C bus connector to query devices. Since there is a mixtures of 2 type of PCF8574 chip on slaves, I will make 2 arrays for the predefined addresses and put them into the sketch!
Again, thank you so much for your time and valuable efforts.
Yours very faithully!
Dang Dinh Ngoc

RobTillaart, 2021-07-22:
You can just put 16 different addresses in one array.
Please let me know if it succeeded (as that is the power of open source)

paulsteigel, 2021-07-22:
Hi Mr. Rob,
There is one point I need your guidance! Should I use Interrupt to trigger the reading of a slave if a pin's state changed (From LOW -> HIGH) as TTP223 module will provide a HIGH signal on the I/O (connected to P0) when touched! Or just better trigger the timer to read every slave (with valid address and online) every 50 milliseconds?
I am a bit struggling with interrupt as the INT pull the line to LOW at powering up and caused ESP8266-01S to fail on booting up.
If interrupt is a better choices, I will find a walk-around on avoiding the INT line to be pulled low on powering up.
Thanks for your further information.
Ngoc

paulsteigel, 2021-07-22:
Hi Mr. Rob Tillart, 
I successfully installed the Interrupt line connected to the GPIO2 (GPIO0 can be okay as well), with a small tweak to allow the ESP8266-01S boot up normally.
[image]
It's a bit late for now, I will try the interrupt procedure from your code tomorrow!.
Thank you a ton for all your attention!
Ngoc

RobTillaart, 2021-07-31:
Sorry for late response, 
Good to hear it all works, I close the issue.

closed 2021-07-31 by RobTillaart