RobTillaart/PCF8574 #28: Platformio folder structure

spassignat, 2021-11-01:
Hello,

I'm suddenly have a build error of this library. Platformio tries to buid test folder (at least) and does not find ArduinoUnitTests.h files.

Reading the platformio documentation, it seems to expect the header files in subfolder include and sources in src. Otherwise it will build everything.

I tried to move the PCF8574.h in PCF8574.cpp in these filders (in the .pio/libdeps) and it just build again like a charm.

There are maybe tips and tricks I don't know about platformio, librairy.properties library.json, ... but would that make sense to have these folders in the project ?
thanks

RobTillaart, 2021-11-01:
All my libraries follow the (old) Arduino guidelines which have the .cpp and .h in the root folder. The separate src folder is only mandatory when different platforms have their own set of code.

Wrt testing I use a GitHub workflow for testing which apparently has conflicting requirements. Never wrote tests for the platformio unittest in any of my libs.

Q1: are there actual problems in the library that blocks your projects?

At the moment I am upgrading the build workflow in all the libraries. I will look if and how I can move sources to the src folder. Will give it a try in this lib later today. 
Would you be able to do some testing as I do not use platformio (only pushed my libs there on request)

RobTillaart, 2021-11-01:
Read the platformIO specs and the .h / .cpp file are OK in the root folder but it is recommended in a src and include folder. 

development branch created to get github build-CI updated first.

RobTillaart, 2021-11-01:
build-CI updated merged in master - will not be a new release as functional there is no change

RobTillaart, 2021-11-01:
Created a **pio** branch to check how much work it would be.
Get failures all over - mainly with include paths.

Conclusion: 
This will be quite some effort and it would break builds resulting in lot of questions / new issues.
So for me moving to src / include is a no go.
(Given that I would need to do this for ~130 libraries in the end => this would mean weeks or even months of effort with no gain - just work imho)

spassignat, 2021-11-01:
Thanks Rob,
I understand, that's kind of big stressful code review. I may had an upgrade of platformio or something else outside of my sources which cause this, I also have similar issue with another lib. (Even going back in versioning I got the same error)

So I think I'll duplicate the source in the project to check, then maybe I'll fork the project in github. More or less only 

(Anyway, thank's for the lib)

spassignat, 2021-11-01:
Overriding src and srcFilter properties in the library.json solve my issue. I don't know if this will help you.
```{
  "name": "PCF8574",
  "keywords": "I2C, PCF8574, PCF8574A, IO, shift, rotate, readButton, toggle, reverse",
  "description": "Arduino library for PCF8574 - 8 channel I2C IO expander, implements shift rotate.",
  "authors": [
    {
      "name": "Rob Tillaart",
      "email": "Rob.Tillaart@gmail.com",
      "maintainer": true
    }
  ],
  "repository": {
    "type": "git",
    "url": "https://github.com/RobTillaart/PCF8574.git"
  },
  "version": "0.3.2",
  "license": "MIT",
  "frameworks": "arduino",
  "platforms": "*",
  "srcDir": ".",
  "srcFilter": "-examples -test"
}
```

RobTillaart, 2021-11-01:
> So I think I'll duplicate the source in the project to check, then maybe I'll fork the project in github. More or less only

That is always an option.

RobTillaart, 2021-11-01:
> Overriding src and srcFilter properties in the library.json solve my issue. I don't know if this will help you.

The github build system uses library.properties (from the arduino lib mgr) for metadata.

spassignat, 2021-11-03:
Never mind, my build options were wrong I was using --target Z_DUMMY_TARGET. Using Production solved my errors.

closed 2021-11-03 by spassignat

RobTillaart, 2021-11-03:
Good to hear it has been solved, thanks for letting me know!