RobTillaart/PCF8574 #40: README doesn't mention interrupt feature

thijstriemstra, 2022-11-15:
I found out by checking the examples.

RobTillaart, 2022-11-16:
Thanks for the issue, I will add a section about it.
This might take some time.

RobTillaart, 2022-11-16:
Added section in the readme - will create a new release asap/

thijstriemstra, 2022-11-16:
awesome, thanks!

> The library does not support this internally.

was added to the readme, but I'm a little puzzled what it means (as opposed to.. externally..?) perhaps it could be clarified

RobTillaart, 2022-11-17:
Ok, i understand the word internal does raise more questions so in some next release I will rephrase.

What I meant to say is that the library cannot handle the pcf8574 interrupts as it has no code for it. The user should catch it in his own code.

RobTillaart, 2022-11-17:
New text is added to master branch, no new release for now.

closed 2022-11-17 by RobTillaart