RobTillaart/PCF8574 #42: Can't change Wire instance on ESP32

thijstriemstra, 2023-03-24:
it's possible to pass a Wire instance to the constructor:

```c++
PCF8574::PCF8574(const uint8_t deviceAddress, TwoWire *wire)
{
  _address    = deviceAddress;
  _wire       = wire;
  _dataIn     = 0;
  _dataOut    = 0xFF;
  _buttonMask = 0xFF;
  _error      = PCF8574_OK;
}
```

but this is overwritten later on when using ESP32 or ESP8266:

```c++
#if defined (ESP8266) || defined(ESP32)
bool PCF8574::begin(int dataPin, int clockPin, uint8_t value)
{
  _wire = &Wire;
  if ((dataPin < 255) && (clockPin < 255))
  {
    _wire->begin(dataPin, clockPin);
  } else {
    _wire->begin();
  }
  if (! isConnected()) return false;
  PCF8574::write8(value);
  return true;
}
#endif

bool PCF8574::begin(uint8_t value)
{
  _wire->begin();
  if (! isConnected()) return false;
  PCF8574::write8(value);
  return true;
}
```

I'd like to be able to use Wire1 (which I passed into the constructor) on an ESP32.

thijstriemstra, 2023-03-24:
Sorry, that `begin(int dataPin, int clockPin, uint8_t value)` is optional and it'll use `begin(uint8_t value)` instead. c++ is still a hobby for me ;)

closed 2023-03-24 by thijstriemstra

RobTillaart, 2023-03-24:
Is your question solved?

Not tried yet but I think you can comment  the _wire assignment, so you can choose the SDA and SCL pins for Wire1

```cpp
#if defined (ESP8266) || defined(ESP32)
bool PCF8574::begin(int dataPin, int clockPin, uint8_t value)
{
  //  _wire = &Wire;   <<<<<<<<<<<<<<<<<<<<<<<< comment  this 
  if ((dataPin < 255) && (clockPin < 255))
  {
    _wire->begin(dataPin, clockPin);
  } else {
    _wire->begin();
  }
  if (! isConnected()) return false;
  PCF8574::write8(value);
  return true;
}
#endif
```
If you can give it a try, we both learn something ;)

thijstriemstra, 2023-03-24:
> Is your question solved?

Yea it's working, sorry for the noise.

RobTillaart, 2023-03-24:
No problemo