RobTillaart/PCF8574 #46: PCF8574 problem podczas kompilacji

Eugeniusz195810, 2023-12-31:
[image]

RobTillaart, 2023-12-31:
Sorry answered wrong issue (deleted)

-----

My PCF8574 library has no **pinMode()** function as it needs not to be set.

This is different from some other IO expanders e.g. MCP23017 which do have a **pinMode()**.

So just leave out the line and it should work.

RobTillaart, 2024-01-04:
Any progress to report?

RobTillaart, 2024-01-04:
The adafruit pcf8574 library apparently has another interface that does support the pinMode command.
It probably does write a 0 or 1 to the device to simulate a pull up. 
My library has no pinMode command but just uses the write command directly.
Note that the device has no DDR ==DataDirectionRegister.

If the adafruit lib works for you I assume your problem is solved and you may close the issue.

Eugeniusz195810, 2024-01-05:
[image]
[image]

Eugeniusz195810, 2024-01-05:
[image]

closed 2024-01-05 by Eugeniusz195810

RobTillaart, 2024-01-05:
Have a look at the examples to see how to use the library.

Thanks for trying.