RobTillaart/PCF8574 #47: Error when using the library with ESP32

HerediaSM, 2024-01-04:
Hello!
I can't get the library to work with an ESP32. I get the following error in the serial monitor:

**[E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263.**

I'm using the example provided for ESP32.
What would be the correct initialization with ESP32?

Thanks in advance.

RobTillaart, 2024-01-04:
Thanks for reporting, 
I will look into it later, will take a few days.

As there are different Esp32, can you tell which one exact ? Might help to recreate the problem.

RobTillaart, 2024-01-04:
The sketch is tested in build-ci 
Which version of the Esp32 toolchain do you use?
Are there similar problems with the other examples?

RobTillaart, 2024-01-04:
Google told me error 263 is a timeout.
This can either indicate bad wiring or incorrect address.
If you use an i2c scanner, which address does it report for the PCF8574?

HerediaSM, 2024-01-07:
The model is ESP-WROOM-32. I was able to solve it using another example.

HerediaSM, 2024-01-07:
The model is ESP-WROOM-32. I was able to solve it using another example.
Also, on the scanner I was getting the I2C addresses, but the wires were failing.

Thank you.

closed 2024-01-07 by HerediaSM