RobTillaart/PCF8574 #50: PCF8575  Latched Input change?    If the main-program, that work with FLAG or Counter from Interrupt, is to slow

Torsten61, 2024-02-04:
Can I use the PCF8575 to reliably record short push-button commands to GND? Even if the program calls the read16() command more slowly.   I use 16 buttons and the Arduino works at limit ;-)
It is not possible with the standard interrupt with flag- or counter-setting, as the current input state is always read out later when reading with read16().
**Read16() is always a snapshot when reading.**

Is there no latch input function so that the cause of the interrupt can be detected?
I only know the possibility to use read16() in the interrupt routine and to store between, which is unattractive prog-style!?

**I once read that the PCF8575 only resets the interrupt trigger when it is read out. But I don't know a SW-solution for this.**

RobTillaart, 2024-02-04:
Thanks for reporting the issue through Github!

RobTillaart, 2024-02-04:
Checked the datasheet and the PCF8575 only has an OLAT = Output LATCH to hold the state.
Also checked the datasheet of the MCP23017 and it does not support input latching either.

So you need to add extra electronics that hold the state.

You might consider using a MCP23S17 (SPI version) which is faster than the PCF8575 (SPI need 4 IO lines)
Expect this one can be read under interrupt.

RobTillaart, 2024-02-04:
Have a look at SX1509 = 16-channel I2C GPIO with LED Driver and Keypad Engine

- https://cdn.sparkfun.com/datasheets/BreakoutBoards/sx1509.pdf
- https://github.com/sparkfun/SparkFun_SX1509_Arduino_Library

It can read a keypad so it should have some input latch

RobTillaart, 2024-02-09:
@Torsten61 
Is your question answered?
If so you may close the issue.

Torsten61, 2024-02-09:
Thanks and close

closed 2024-02-09 by Torsten61