RobTillaart/PCF8574 #51: Need help with switch

souvikch, 2024-03-01:
Hi,
I have gone through the documentation avalible over internet but I'm unable to get input with PCF8574. I have tried with buttons, but that's not working even the exaple codes. The hardware I'm working with is mentioned below

Wemos D1R1 Mini(ESP8266)
Arduino IDE

RobTillaart, 2024-03-01:
A good tutorial - https://pu2clr.github.io/PCF8574/

You need to connect the button / switch typical between GND and PCF-IO pin.
And from the PCF-IO pin you need a pull up resistor to 5V 

When the switch is not pressed you will read HIGH for that pin.
When the switch is pressed the GND will draw the level to LOW.

----
you can reverse polarity by having the switch between the IO pin and 5V
and a pull down resistor between the IO pin and the GND.

souvikch, 2024-03-02:
Thanks for your quick response. I'm working on it and initial tests are
good. I'm also working on interrupts on PCF8574 and I'll be back to
you soon.

Thanks
Souvik

RobTillaart, 2024-03-10:
@souvikch 
Any progress ?

souvikch, 2024-03-13:
> As no progress is reported I assume the issue is solved. Feel free to
> reopen if needed.

Yes, my problem with switch is solved but I'm still not able to get the
interrupts and I'm still working on that. But in between in stuck with
other work (my job related) and not able to give time to this. However I'll
restart again this weekend.

Thanks

Souvik