RobTillaart/PCF8574 #52: Can't use setAddress(x) with begin() since 0.4.0

davidrule1969, 2024-10-05:
Here is a rough snippet of my code.  It works with 0.3.9 but not with 0.4.0 or 0.4.1.  I've read the note regarding breaking changes with 0.4.0 but can't see what I am doing wrong.

```
void connect_to_pcf8574() {
    PCF8574 pcf8574;
    pcf8574.setAddress(0x20);
    if (pcf8574.begin()) {
      // do stuff
    }
}
```
With 0.3.9 the _pcf8574.begin()_ function returns _true_ but with 0.4.0 and 0.4.1, it returns _false_.

RobTillaart, 2024-10-05:
Thanks for the issue, sounds like a bug.
will look into it later today

RobTillaart, 2024-10-05:
Can you check what **pcf8574.setAddress(0x20);** returns?

Both begin() and setAddress() call **isConnected()** under the hood.

RobTillaart, 2024-10-05:
Have you called **Wire.begin()** in your setup(), 
or at least before the call to ```void connect_to_pcf8574()```
Because that is what needs to be done since 0.4.0

davidrule1969, 2024-10-05:
> Have you called **Wire.begin()** in your setup(), or at least before the call to `void connect_to_pcf8574()` Because that is what needs to be done since 0.4.0

That fixed it!  I did read the note about needing to call **Wire.begin()** but incorrectly assumed because I was not setting the pins in **begin()**, that I did not need it :-(

Thanks for your guidance in getting this working.  I very much appreciate it :-)

RobTillaart, 2024-10-05:
Good to hear it is solved, will close the issue.

closed 2024-10-05 by RobTillaart