RobTillaart/PCF8574 #53: Question: Pullup for pins

hasenradball, 2024-11-03:
Hi,

wanted to ask about the pin behaviour of the chip.
Can you set a internal pullUp to have a defined state?
Or how is it realized?

Frank

RobTillaart, 2024-11-03:
Thanks for your question,

From my head, you need to check datasheet for a definitive answer.

There is no explicit internal pull up command otherwise I would have added this long ago.

You can write a 1 to an input pin and then read it back. If the line is floating it will return 1, if connected to GND it will return zero. You should be able to verify this easily.

In my projects I prefer explicit pull up / down as that gives more control over the edges of pulses (think long lines). 

Does this answer your question

hasenradball, 2024-11-03:
Yes perfekt thank you!
😊

closed 2024-11-04 by RobTillaart

hasenradball, 2024-11-06:
@RobTillaart,

may I ask on question to your lib?
Do you think it is possible to readout an AM2302 Sensor(aka DHT22) by the PCF8574?