RobTillaart/PCF8574 #54: Comment functions

hasenradball, 2024-12-02:
Hi Rob,

did the first version may be could you give some input for the `???` parts.
A short review of the comments would be very great!.

See here:
[first changes](https://github.com/RobTillaart/PCF8574/compare/master...hasenradball:PCF8574:master)

RobTillaart, 2024-12-02:
A quick look
- looks very doxy
- why replace line comments with block comments  //  ==>  /****/   ???
- the comments so far add no or only little value above the function names. 
- reverse ??? => explained in readme.md
- button mask ??? ==> explained in readme.md
- read button ??? ==> explained in readme.md
- select ??? ==> explained in readme.md
- selectN ??? ==> explained in readme.md
-

hasenradball, 2024-12-02:
What do you mean by this?

the comments so far add no or only little value above the function names.

Not sure what you mean, exactly.

I do type 
/**/
then enter in vscode

and then the closure fore doxy comment is created automstically. To be able to describe parameters an returns.
These are the used by doxygen.

RobTillaart, 2024-12-02:
E.g. this header change, it is not a showstopper but it is not needed

```cpp
/*
 *    FILE: PCF8574.cpp
 *  AUTHOR: Rob Tillaart
 *    DATE: 02-febr-2013
 * VERSION: 0.4.1
 * PURPOSE: Arduino library for PCF8574 - 8 channel I2C IO expander
 *     URL: https://github.com/RobTillaart/PCF8574
 *          http://forum.arduino.cc/index.php?topic=184800
*/
```

replaces 

```cpp
//
//    FILE: PCF8574.cpp
//  AUTHOR: Rob Tillaart
//    DATE: 02-febr-2013
// VERSION: 0.4.1
// PURPOSE: Arduino library for PCF8574 - 8 channel I2C IO expander
//     URL: https://github.com/RobTillaart/PCF8574
//          http://forum.arduino.cc/index.php?topic=184800
```

hasenradball, 2024-12-02:
sure thats not nedded.
Its only cosmetics for the eyes
😊

hasenradball, 2024-12-02:
Just asked for some help an examples…
just for info…
😉

[image]

[image]

[image]

RobTillaart, 2024-12-02:
PS, 
chatgpt is imho not a trusted party, it is great for inspiration to get started but I have no trust in the quality of code it provides. 
Last year I have seen bad code and documentation that lowered the value of projects created by chatgpt. (see also Arduino Forum) and only if it can be argued that new code / documentation adds value it will be included. (So if chatgpt writes decent code that adds value or solves a bug and leaves maintainable code behind, I'm completely fine with it.

hasenradball, 2024-12-02:
> * E.g. if there is a parameter named **deviceAddress** and your popup states **the address of the device**. That is a waste of time both to write it as developer and to read it as user. If it states **the address of the device, between 0x20 and 0x2F** then it adds some value. For more complex stuff one can provide a link to the readme / datasheet.

Totally agree to this hint, sure but I did only draft a first version, which brings up stuff for dicussion.

RobTillaart, 2024-12-03:
I am interested in (doxy comments with) added value, otherwise I would not spend time on the issue.
So please create a PR showing added value with doxy

hasenradball, 2024-12-03:
> I am interested in (doxy comments with) added value, otherwise I would not spend time on the issue. So please create a PR showing added value with doxy

I am on it your library is worth to be enhanced 👍.

But please can you give me a short hint about read/write and Button functionality. Did not catch it from the Readme.

I use e.g. a PCF where I read value on Pin0 and drive 3 LEDs at Pin1...3.
I use write and read for Pin0, and also to set the Pins for the LEDs.
Is it wrong?

RobTillaart, 2024-12-03:
> But please can you give me a short hint about read/write and Button functionality. Did not catch it from the Readme.

The button functions were added by someone else, never used them myself.
From my head the goal is to mask input / output if you have mixed IO in one PCF8574.
I need to reread the docs/code to get the details again.

hasenradball, 2024-12-03:
Would it be ok for you when I put the file description in doxy style like

````
/**
 *    @file PCF8574.cpp
 *  @author Rob Tillaart
 *    @date 02-febr-2013
 * @version 0.4.1
 *   @brief Arduino library for PCF8574 - 8 channel I2C IO expander
 *     @url https://github.com/RobTillaart/PCF8574
 *          http://forum.arduino.cc/index.php?topic=184800
*/
```

RobTillaart, 2024-12-03:
OK, give it a try, should do no harm.

RobTillaart, 2024-12-03:
peeked the code:
The button functions write a HIGH to the INPUT pins. So they are working as a sort of pull up.
This will return HIGH on INPUT pins that are not switched to GND.

(back to my project, need to finish some things this week, so answers may take a while)

hasenradball, 2024-12-03:
Here the first try

[refman.pdf](https://github.com/user-attachments/files/17991455/refman.pdf)

hasenradball, 2024-12-03:
Here the first try...

[refman.pdf](https://github.com/user-attachments/files/17991520/refman.pdf)

[doxygen_log.txt](https://github.com/user-attachments/files/17991528/doxygen_log.txt)

hasenradball, 2024-12-03:
As I looked into the code for documentation, I found some points I would suggest to improve in code.

a) make `getAddress()` a const function => does not change the object

b) improve constructor speed and put private members in descending order of type to optimize memory

```

//old
PCF8574::PCF8574(const uint8_t deviceAddress, TwoWire *wire)
{
  _address    = deviceAddress;
  _wire       = wire;
  _dataIn     = 0;
  _dataOut    = 0xFF;
  _buttonMask = 0xFF;
  _error      = PCF8574_OK;
}

//new
=> cpp file:

PCF8574::PCF8574(const uint8_t deviceAddress, TwoWire *wire) : _address{deviceAddress}, _wire{wire}
{}

=> header file:

	private:
          int _error {PCF8574_OK};
	  uint8_t _address;
	  uint8_t _dataIn {0};
	  uint8_t _dataOut {0xFF};
	  uint8_t _buttonMask {0xFF};
	 
```

c) use Namspace for PCF8574

put all stuff into a Namespace like for example done here:
[AM2302](https://github.com/hasenradball/AM2302-Sensor/blob/master/src/AM2302-Sensor.h)

hasenradball, 2024-12-03:
here the latest pdf from actual state:

[refman.pdf](https://github.com/user-attachments/files/17993274/refman.pdf)

I see you also provide a MCP23008 lib.

RobTillaart, 2024-12-03:
> As I looked into the code for documentation, I found some points I would suggest to improve in code.

Should be a new issue imho.

RobTillaart, 2024-12-03:
> I see you also provide a MCP23008 lib.

yes, written about 270 libraries, good bad and ugly :)

RobTillaart, 2024-12-03:
> b) improve constructor speed and put private members in descending order of type to optimize memory

Tested on AVR UNO, compiler did not save any byte, neither in code or global variables.
So the added value is not memory. Speed could save a few microseconds.

hasenradball, 2024-12-03:
> improve constructor speed and put private members in descending order of type to optimize memory

Could be but why not sorting things in the right way if observed ;-)

closed 2024-12-03 by hasenradball